gzweigle/open-hybrid-piano #6: USB MIDI?

davidedelvento, 2023-12-10:
Great to see that you included DIN-MIDI. It's not clear if the now-more-popular USB-MIDI is supported already. If not, it should be relatively easy to add, since the Teensy 4.1 can act as a USB Host. I did that with the [pico](https://github.com/davidedelvento/Mybrid/tree/main/RaspberryPiPico) using https://github.com/hathach/tinyusb so it should be similarly easy on the Teensy (if not done already -- if done already, this "bug" is mentioning it on the main README)

gzweigle, 2023-12-10:
I wish Teensy was not host.  But yeah good suggestion for future.

gzweigle, 2024-09-02:
I added to the IPS 2.1 enhancement list.
Closing this issue since it is now a duplicate

closed 2024-09-02 by gzweigle