gzweigle/open-hybrid-piano #7: Pedal options?

davidedelvento, 2023-12-10:
The README mentions the option to use external pedals with 1/4" connections, which is great.

Some people (like me :wink: ) may would like to use an existing acoustic pedal, with a sensor similar/identical to the ones for the keys/dampers.

Looking at https://github.com/gzweigle/DIY-Grand-Digital-Piano/tree/main/packages/StemPianoG#pedal-connections it may seem that this is already possible, if my understanding is correct. If so, update the main README accordingly, specifying this option, otherwise let's discuss here what it would take to make it possible

gzweigle, 2023-12-10:
StemPianoG code does not support (yet) but hardware does support. So, I will leave here as an enhancement idea, then update documentation if code is modified in future.

gzweigle, 2024-09-02:
code supports, if needs modification for a specific pedal model will open a new issue for that case.

closed 2024-09-02 by gzweigle