gzweigle/open-hybrid-piano #13: repetition false strike

gzweigle, 2024-01-13:
repro:
Press piano key quickly.
Slowly release key until jack moves beneath knuckle.
Hammer moves up.
On an acoustic piano, hammer does not hit string.
With stem piano sometimes get a false strike.

soln:
Probably can fix with firmware and might also need more consistent sensor placement.

davidedelvento, 2024-01-14:
> With stem piano sometimes get a false strike.

I haven't looked at your code, but with other codes I've played with (mine and Jay's), this is the first basic calibration that one needs to do! That's the advantage to have a mostly software (unlike Cybrid which is mostly hardware) design.

You have to play slowly (actually could even just raise the hammer manually rather than with the key) without hitting the string/sensor. That needs to be calibrated as "let off" position. Then you have to have the hammer hitting the string/sensor and calibrate that as "play" or something. You need to do that for each and every key because both the sensors and their placement will be different from one key to another. On Cybrid, you do that with the potentiometers (huge pain), with Mybrid and Jay's piano-conversion (and I assume you can do that with yours) you write a routine in which you ask the player to play each and every key first silently then not-silently and use those 88+88 values from the ADCs as calibration values. A fancy calibration GUI/script may allow to recalibrate only some notes rather than everything or globally shift everything up or down etc.

davidedelvento, 2024-01-14:
See also: https://github.com/jkominek/piano-conversion/wiki/Calibration-Procedure

gzweigle, 2024-04-20:
this seems to have been fixed by other changes. will close for now and reopen if happens again.

closed 2024-04-20 by gzweigle