gzweigle/open-hybrid-piano #14: note consistency

gzweigle, 2024-01-13:
repro:
Play multiple notes with same velocity. For example, using a ruler.
MIDI value is not exactly same for all notes.

soln:
Probably sensor consistency.
ADC - differential?

davidedelvento, 2024-01-14:
Again, https://github.com/jkominek/piano-conversion/wiki/Calibration-Procedure has some ideas on this. One could make the calibration procedure I described in #13 also ask for a fff strike and use that to calibrate the slope of the velocity computation. That is a bit tricky because while it's easy to raise the hammer to let-off without it going much higher, and it's also easy to have the hammer strike the sensor/string with "whatever" velocity, it's much harder to make it strike it with consistent velocity. I speculate (and my preliminary tests I described in the Mybrid repo confirm) that this calibration is not necessary once one has calibrated the let-off and the strike point and hence measured velocity only between these two values. But then I only had a handful of notes working, never a full set as you do, so I might be missing something, either on the user "feeling" side or on the sensor variability side. Also, my sensors different than yours and are probably mounted with a bit tighter tolerances, as you have probably already seen at https://pianoclack.com/forum/d/542-update-on-piano-conversion/

gzweigle, 2024-01-15:
I'm in the process of checking in details of a system I built to accurately measure sensor response.

gzweigle, 2024-04-20:
improved by a combination of rebuild/regulate action, redrill sensor bar with more accuracy, and real-time algorithm in firmware (code is in repo now)

closed 2024-04-20 by gzweigle