gzweigle/open-hybrid-piano #15: power supply noise

gzweigle, 2024-01-13:
repro:
Detailed measurements show quite a bit of 120 Hz noise.

soln:
Better +5V power supply.
Look at grounding.
Differential ADC.
Filters.

davidedelvento, 2024-01-14:
One of the most notably effect of power supply noise is on the intensity of the LED light, which in turns would cause a change of measured distance (unless it's very carefully calibrated, which IMHO is a big pain) with that frequency. That could turn out as a subtle modulation of the velocity at that frequency, since when we are in a peak intensity the hammers would appear to be closer (and hence hit the string/sensors earlier, and hence have a higher velocity/loudness) and the reverse when we are in a trough with hammers appearing more distance and etc-etc lower velocity/loudness.

See https://github.com/jkominek/piano-conversion/wiki/LED-Power-Board for a solution to this problem, which also has the added benefit to removing the needs of all those current-limiting resistors which are needless power sink. Not much each, but even a fraction of a watt when multiplied by 176 becomes non-negligible (that explain the lack of such resistors in my sensorboards as we were discussing elsewhere)

gzweigle, 2024-01-21:
Turns out this was due to ambient room lighting shining on the sensors.
With sensors shielded, the 120 Hz noise goes away.

closed 2024-01-21 by gzweigle