gzweigle/open-hybrid-piano #21: Video for "Piano Action Frame" is set to private

WebSnke, 2024-02-20:
Hi, I am a huge fan of this project and would love to build my own. However, the video for building the piano action frame is set to private.
Is this intentional or just a mistake?

closed 2024-02-20 by WebSnke

davidedelvento, 2024-02-20:
Duplicate of #20

gzweigle, 2024-02-22:
https://github.com/gzweigle/DIY-Grand-Digital-Piano/discussions/22

WebSnke, 2024-02-23:
> Duplicate of #20

Yeah, noticed that too...
Btw, has anyone managed to put this into an upgright piano action?

davidedelvento, 2024-02-24:
> has anyone managed to put this into an upgright piano action?

That would be better for another issue or a discussion item e.g. on https://pianoclack.com/forum/ 

Hope Greg don't mind too much for me answering here instead. I don't think anybody has done that. In a nutshell, upright pianos have the following limitations:

* the action is harder to remove (not extremely hard, but harder than grand) and when removed, unless you have specialized equipment, you are guaranteed to break it because it you cannot lay it in any position without having the whole weight pushing on a brittle part, unlike in a grand where you can just pull it out and place it on any table
* the keyboard is part of the cabinet and cannot be removed, if not one key at the time (and then you need to rebuild a keyframe), unlike in a grand where the keyboard is attached to the action
* if you want to reuse the upright cabinet you may want to remove the harp to make it lighter. This is close to impossible (unlike in a grand, where it is easy if you have a hoist, and demanding but not too hard if you work with *several* people in the operation)
* if you don't remove the harp, you have to remove all the strings, which is a very long, tedious and dangerous job -- at that point you have very limited working space for the sensors. Again, not impossible, but much harder than a grand
* assuming that you succeed in all of the above, you end up with an arguably inferior action than a grand, after having done much more work -- hence I don't see why one would embark in such an endeavor. 
* anticipating an answer of "but I don't have room for a grand in my house", my comment is "me neither" and in fact I made an "upright" (actually a spinet) out of a grand action. That was EASY. See https://pianoclack.com/forum/d/170-build-your-own-hybrid-piano-step-two-the-cabinet for details.

WebSnke, 2024-02-24:
Thanks for your response. Let's continue this on the forum.