gzweigle/open-hybrid-piano #43: HW: location of sensor resistor

davidedelvento, 2024-05-05:
I am wondering if you have considered putting the sensor resistor after the multiplexers, so you would not need to have (solder) 88 of them. 
In my case that's obviously because I don't want to modify my sensorboards which have been already fabricated without, and I don't want to massively change your main board either with 88 resistors. Obviously I could have them in the cables and that might be my chosen route, but putting them after the multiplexer seems also doable, so asking if you considered that (and perhaps discarded it for some reasons).

gzweigle, 2024-05-05:
I ran SPICE simulations of various options last year.
Settling time (to ADC resolution) limits the sample rate of every option.
So I need the circuit to settle as quickly as possible.
CNY70 trying to drive through multiplexers slowed things down too much.
An amp (ie TIA) improves the no-resistor settling time problem but only if before the multiplexers, then need 88 amplifiers.
Key goal of stem piano G is lowest possible cost.

However, stem piano G is pretty much done. Maybe next piano should be with amps vs. R.

For reference, these diagrams show the ADC timings, although not the settling time simulations:
https://github.com/gzweigle/DIY-Grand-Digital-Piano/blob/main/design/data_acquisition/eca0X_ips2X/eca0X_ips2X_timing_design.md

davidedelvento, 2024-05-05:
Yes, that's exactly what I had in mind, but just as "gut feeling". So with current design it is better to put the resistors in between the multiplexers and the sensors.

> An amp (ie TIA) improves the no-resistor settling time problem but only if before
> the multiplexers, then need 88 amplifiers. Key goal of stem piano G is lowest possible cost.

Well, using TIA is not too bad, many IC have several op amps in the same package and cost only pennies. FWIW, that's what https://github.com/jkominek/piano-conversion/ intend(ed?) to use see https://github.com/jkominek/piano-conversion/wiki/Analog-Stage-Theory-of-Operation#transimpedance-amplifier
In fact I might for intermediate board(s) on which I will mount resistors for now and which I will replace with a TIA later. 

Maybe for the next version of the sensorboards you may do the same so you will increase flexibility (i.e. not require replacing 88 expensive sensorboards) if one decides to switch between options.

Thanks for the discussion!

gzweigle, 2024-05-05:
Thanks. Agree.

closed 2024-05-05 by gzweigle