gzweigle/open-hybrid-piano #44: FW: option to use optical sensors for pedals

davidedelvento, 2024-05-05:
In your design you have 3 jacks for external pedals. In a full DIY grand (such as the one I have built), one may want to use the original pedals and use the same optical sensors (properly mechanically adjusted for the pedal travels) instead. Since the design already has 96 channels and most pianos have 88 keys, it should be easy to allocate 1, 2 or 3 channels to the pedals.

gzweigle, 2024-05-05:
Yes that will work.
Cases like that are exactly why I put header J46. Can bypass the jacks for up to 8 pedals.
See pedal section of:
https://github.com/stem-piano/stem-piano-g-main/blob/main/documentation/assembly_manual.md
Skip the wires, connect direct to J41, 43, 45.

davidedelvento, 2024-05-05:
Oh, I missed that, it's great!
So it's already supported in firmware?

gzweigle, 2024-05-05:
Just minor firmware changes.

closed 2024-05-05 by gzweigle

davidedelvento, 2024-12-15:
I attempted to get this to work in the most naive way, and I went nowhere. Before I spend too much time going in the wrong direction, and reverse engineering what you have done, can you please nudge me towards what you had in mind here?

My pedal sensors are identical to your key sensors, so I need the three-way connector (I see that in the wiring diagram you connect only two). Also, it (wrongly?) appears to me that you wire two ADC channel per pedal, but one of such channel is just a "flag" and currently not in use. Is that right?

TIA!

gzweigle, 2024-12-15:
Yes the pedal jacks have two outputs.
One output is the pedal analog signal.
The other output is an optional indicator "flag" that the pedal is connected. 

On the IPS 2.0 circuit board, each output goes to a separate ADC channel.
And, yes, correct, the pedal connection indicator "flag" is not used by the firmware.

Connecting a pedal directly to the header at the back of the IPS 2.0 circuit board should work.  Since all the pedal jack is doing is writing to one of the ADC channels.

davidedelvento, 2024-12-16:
Thanks, so to confirm: 

1. I don't have to change anything in the multiplexer code, that channel is already read by the default multiplexing settings of the firmware
2. The power is also properly connected so if I attach a key sensor (equivalent to your HSB, not mine which has external power) to the pedal channel, that key should act as a pedal -- and if I don't see any MIDI pedal message that's a problem with either wiring or sensor (not firmware)

Correct?

gzweigle, 2024-12-16:
(1) - There is a setting for which ADC channel routes to the pedal algorithm in the firmware.  The setting allows any of the 96 inputs to be used for pedal(s).
https://github.com/gzweigle/open-hybrid-piano/blob/main/firmware/releases/ips2_hammer/hammer_settings.cpp

(2) - The HPS sensors require 3.3V and ground connections so those two of the three IPS/HPS pins are "outputs" to the HPS. If your pedal gets power by another means, then only one connection is required from the pedal to the IPS board.

Note that pedals with a passive variable resistor will need some way to bias and set the proper output voltage to the ADC.   This is the purpose of the pedal level shift (PLS) card.
https://github.com/gzweigle/open-hybrid-piano/tree/main/hardware/releases/pls00

davidedelvento, 2024-12-16:
(1) perfect thanks

(2) I must not have explained myself -- My pedal is not a digital piano pedal (neither active nor passive). My pedal is the original pedal from the acoustic piano I took the action from! I have installed an HPS into it. As such it does need power and ground. 

Given what you wrote in (1) I assume that from the hardware perspective those additional pins are correctly wired to power and ground and if my pedal is not working (which is the case) something else in the wiring or in the sensor must be wrong. Next weekend I will start debugging from there

Thanks!

davidedelvento, 2025-01-04:
Is it supposed to output continuous values? Looking at the code I'd say yes, but....

I managed to get it to work, but it outputs only 0 or 127 on CC64 (sustain). Have you tried hooking one of the optical sensors and verified it gives intermediate values too?