gzweigle/open-hybrid-piano #47: FW: cal rare outlier

gzweigle, 2024-05-11:
Get an outlier that breaks calibration for a single note.
Happens very rarely.

davidedelvento, 2024-05-11:
What do you mean by "breaks calibration"?

gzweigle, 2024-05-11:
The firmware is now automatically compensating for sensor nonlinearity.

Its a huge improvement over the problems I described in my last YouTube video.

(for my skill level) the piano "regulation" is now very good.  And, I have ideas to make even better too.

However, on rare occasions a note will go out of calibration. Then that note sounds slightly louder or quieter than others. My guess is: either a firmware bug or unexpected/unknown noise.

Code link:
https://github.com/gzweigle/DIY-Grand-Digital-Piano/blob/main/firmware/releases/StemPianoIPS2/src/calibration_sensor.cpp

davidedelvento, 2024-05-11:
Ah, very nice! I am preparing/purchasing what I need to finally solder/assemble one (still need to figure out the PCB order), and I hope that by the end of June I will be able to contribute something more specific to (minor) issues like this!!

gzweigle, 2024-05-11:
cool.
I need to get a few very minor HW changes in.

Also, slowly migrating to a GitHub organization https://github.com/stem-piano/

Going slow because of all the links to original.

gzweigle, 2024-08-10:
Can't repro

closed 2024-08-10 by gzweigle