gzweigle/open-hybrid-piano #52: HW: new IPS 2.1

gzweigle, 2024-05-13:
Presently working on a new iteration of the IPS mainboard.
Goal: all changes compatible with existing IPS 2.0 firmware.

Moving CAN connector to rear of board.
Adding another LED/testpoint to front by Teensy (making TFT IRQ optional with a jumper resistor).
Putting extension SCA power on main 3.3V instead of an LDO 3.3V.
Move J46 to right.

Investigating adding 96 optional 805 surface mount resistors to bottom side of board in support of another sensor approach:
https://github.com/stem-piano/sensor-systems/tree/main/qps

Addressing these existing issues:
(1) Adding ground pin to J46
https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/50
As optional gnd connect point for new PLS board: https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/26
(2) Adding SCA connector labels
https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/41
(3) BOM - changing LDO part
https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/45

davidedelvento, 2024-05-13:
Awesome!!
If it's not too many irons in the fire, maybe investigate adding some TIAs. If they need to be before the mux (and hence be 96 of them) perhaps it's better to make an interposer board. If they can be after the mux (and hence just a hanful) putting them on the IPS makes more sense. We'd need to run some simulation (if you haven't already)

gzweigle, 2024-06-08:
> perhaps it's better to make an interposer board

Good point.  Will use the 96 sensor headers.  Created issue: https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/53

Therefore:
- Removing the 96 805 surface mount resistors from scope of IPS 2.1.
- These (and optional TIA) will be on the new removable sensor interface boards.  

This means that, while IPS 2.1 has some nice improvements, its not necessary.  The existing IPS 2.0 is ok.  So, likely I will lower the priority of creating IPS 2.1.

gzweigle, 2024-06-15:
Updated list of changes from IPS 2.0 to IPS 2.1:
(1) Move CAN connector next to quarter inch jacks.
(2) Add another LED/testpoint to front by Teensy (making TFT IRQ optional with a jumper resistor).
(3) Put extension SCA power on main 3.3V instead of an LDO 3.3V.
(4) Move J46 to right to make room for CAN.

(5) Add ground pin to J46: https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/50
(6) Adding SCA connector labels: https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/41

IPS 2.1 is on hold because these improvements are minor.

gzweigle, 2024-06-23:
Another addition for IPS 2.1. Jumper on R23, R25, and R28 so can use active or passive pedals without having to change whether resistors are soldered onto the board.

gzweigle, 2024-07-01:
Another one - for both Teensy and SCA, include the headers in the schematic.

gzweigle, 2024-07-12:
Another one - external connection for a USB MIDI port in addition to the serial MIDI port.

gzweigle, 2024-07-12:
More improvements for IPS 2.1:
The +5V - USB power info printed on the circuit board are overly restrictive.
Could also add information like this on back of board?

davidedelvento, 2024-07-13:
Since you are planning to redo this, in my opinion is better to do as vagfilm mentioned: replace all of these with just headers and let all the jacks to be "floating". This will greatly simplify assembly, allow more flexibility, and provide the option to build a nice enclosure where the jacks can be safely attached without risk of stressing/ruining the PCB by continued use.

gzweigle, 2024-07-13:
Its a good idea to make a version with a smaller PCB, no TFT or quarter inch inputs.

Note that presently all 96 inputs are independently available along the back of the PCB. The quarter inch jacks connect only to header J46 and nothing else. Does that answer the first question?

davidedelvento, 2024-07-13:
> Its a good idea to make a version with a smaller PCB, no TFT or quarter inch inputs.

Agreed. I would go as much as to not have Ethernet, MIDI, power supply jack or anything. But I don't mean that these are not possible, just that they are not directly attached to the board. In fact, I think all of them (including the TFT) would be at least "nice to have" if not "absolutely needed" but the design could be better served if all of them are attached to the board via a ribbon cable and then mechanically held somewhere else (e.g. in the enclosure).

> Note that presently all 96 inputs are independently available along the back of the PCB.
> The quarter inch jacks connect only to header J46 and nothing else.
> Does that answer the first question?

I guess you mean the other ticket? If so, yes. In this issue report I didn't have any question in my ticket only a suggestion (which you are free to consider or reject as you see fit, obviously).

gzweigle, 2024-07-14:
Interesting idea... I could put everything optional, or that could have variants, with cable connectors.

gzweigle, 2024-07-26:
Info on circuit board include to cut 5V pad

gzweigle, 2024-09-02:
Add USB host connection

gzweigle, 2024-09-02:
For USB host - see this discussion: https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/6

technolhodgy, 2024-09-02:
I have been keeping up with you Youtube Video's
This is a great project, and I look forward to seeing it progress.

I picked up a Grand Action last year that was going to be broken down.
I have just about had an octave working on a Pi Pico, and plan to scale it up to 88 notes soon.

gzweigle, 2024-09-07:
Consolidated change list (so far!) to consider for a future IPS 2.1 board spin, along with a few additional items:

Features:
USB MIDI (I think it has to be host?) connection.
Quarter inch jack interposer board instead of presently forcing specific quarter inch jack types.
Push button in addition to switches.
Dedicated TFT processor with fast command-based processor2processor interface.

Improvements:
Move CAN connector to back of board.
J46 – move to make room for CAN.
Add another LED/testpoint to front by Teensy (making TFT IRQ optional with a jumper resistor).
Put extension SCA power on main 3.3V instead of an LDO 3.3V.
J46 – add ground pin.
Jumper on R23, R25, and R28 so can use active or passive pedals through the quarter inch jacks without solder change.
Socket the fuse.
TFT LDO.
Move Teensy forward

Labels:
SCA connector labels.
For both Teensy and SCA, include the headers in the schematic.
New +5V - USB power info on circuit board.
Put additional instructions on back silkscreen.

Fixes:
Add external pullups to switches (addressing a Teensy problem).

General:
Could I keep it "99%" compatible with IPS 2.0 codebase?

davidedelvento, 2024-09-08:
> USB MIDI (I think it has to be host?) connection.

- [ ] No why do you want to make host? A keyboard is a device and so this should be.
- [ ] I have other things to discuss but let me open a separate issue for them, not to pollute this one with discussion but leaving it just as a list of things which you will have concluded are worth implementing -- in fact I will delete this comment if you ask me to, so we keep this issue clean
- [ ] speaking of lists, see how you can make it very nicely? Click on the "checklist" icon to the left of the paperclip (you should be able to edit your earlier message)

gzweigle, 2024-09-21:
See if can make space for stickers next to LED.  Then, a user can make it easy to identify the LED function based on the firmware.

gzweigle, 2024-09-28:
Moving the comment from issue https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/94 to this consolidated ips 2.1 changes issue.

"The silkscreen for the IPS should include a text saying if the + is inside or outside"

gzweigle, 2025-05-04:
Consider swapping 3.3v and gnd rows at sensor header to simplify wiring in some cases.
See: https://github.com/gzweigle/open-hybrid-piano/issues/96