gzweigle/open-hybrid-piano #55: FW: improve the velocity algorithm

gzweigle, 2024-06-09:
During testing and creating most recent video on note-to-note consistency, I notice the reported velocity is less consistent than the hammer position. Probably due to the derivative in algorithm. I have a few ideas to improve and need to get into the code.

davidedelvento, 2024-06-10:
IMHO, the main problem here is how to get a "ground truth" on which to assess the algorithm, and then find the root cause to the inconsistencies. Some questions for you to consider (in addition to what you are already thinking about)
* Is single note velocity always "perfect" for some way of considering perfection? \
* Do you still have the single note action which you could use for testing? Can you install two sensors for that same key and see how they differ from each other, ruling out difference in mechanics/regulation from the noticed difference in measured velocity?

gzweigle, 2024-06-22:
Discoveries/Improvements so far:

(1) For slow strike (quiet note) cannot use the max() algorithm because the hammer travel is under tighter control of pianist than when fast strike (louder note). This could explain why sometimes the piano feels inconsistent when playing ppp. Use a window back from strike point. 

(2) Should not use the velocity zero crossing as time of strike because derivative has timing variations susceptible to signal asymmetry. Even though it is linear phase. I have another algorithm.

(3) Tests and data are indicating that the type of foam used for hammer stop may play a bigger role in velocity accuracy than I initially assumed.

None of this in GitHub code yet.

gzweigle, 2024-07-07:
Details of the existing and improved velocity algorithms will be covered in the technical paper:
https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/73

gzweigle, 2024-08-10:
Added a second algorithm.

Algorithm 0: No thresholds. Less dependent on precise mechanical setup. Longer delay. Less accurate.
Algorithm 1: Uses thresholds. Requires a mechanical setup with tight tolerances. Minimal delay and more accurate.
May add more algorithms in future

closed 2024-08-10 by gzweigle