gzweigle/open-hybrid-piano #57: HW: resistor type

gzweigle, 2024-06-16:
A few thick film resistors where thin film is better.

hps06_ips2x_bill_of_materials.txt, hps07_ips2x_bill_of_materials.txt:
CRCW080550R0FKTA (50 ohm thick film resistor) replace with ERA-6AED510V (51 ohm thin film resistor).

hps08/hps08_ips2x_bill_of_materials.txt:
Missing the 180 ohm part number. Add: ERA-6AED181V (180 ohm thin resistor).

sca00/sca00_bill_of_materials.txt:
0.2 ohm is thick film, can find alternate?

davidedelvento, 2024-06-16:
https://www.arrow.com/en/products/rl1632r-r200-f/susumu ? 
It's only 1% and not better as you might have hoped, but still is thin film and perhaps adequate or at least better than you have already?

gzweigle, 2024-06-16:
Thanks, I will try that resistor.
Closing the issue because checked in parts files with new resistor values and also will test the suggested.

closed 2024-06-16 by gzweigle