gzweigle/open-hybrid-piano #66: HW: sca part inconsistence

davidedelvento, 2024-06-30:
The BOM lists `U1` as `REF6025` but kicad lists it as `REF6030`

gzweigle, 2024-07-01:
Good question.
The part comes in variants 2.5V, 3.0V, 3.3V, 4.1V, 4.5V and 5.0V.
Probably when I did the schematic I was thinking 3.0V would work. However, that made VCE across the CNY70 too small.  All my documentation in stem piano repo is for 2.5V.
For your sensors you may be able to use a 3.0V or even 3.3V (check the amp range) as the reference value.  The higher the reference the better for signal.

I added a comment to hardware README files that the bill of materials wins if different than schematic.

closed 2024-07-01 by gzweigle

davidedelvento, 2024-07-10:
I am not sure I understand the problem of `Vce` becoming too small when using an `U1` which provides a higher voltage.

Assuming that `Vce` is `5V` minus what `U1` provides and comparing Fig.3 of https://www.sparkfun.com/datasheets/Robotics/QR_QRE1113.GR.pdf with Fig.8 of https://www.vishay.com/docs/83751/cny70.pdf it appears indeed that `3.0V` might be marginally better. 

But looking at the characteristics tables I don't see any other differences. Given the complexities of asking the vendor to manufacture some boards with one part and some others with another part, I'm tempted to let everyone take the `2.5V` version. Do you think that would be a problem? I guess in the worst case I could try to desolder and replace it if necessary (it would be a painful thing to do, but not impossible, as far as I can tell...)

gzweigle, 2024-07-11:
Yes, for HPS, vdd=3.3 and with vref = 2.5 the difference (vce) is 0.8 volts when hammer is at stop.
If vdd > 3.3, vref can be bigger.
I agree vref=2.5 for all boards is simpler.
Select ACE resistors so the output is 0 to 2.5v as hammer goes from rest to stop.

gzweigle, 2024-07-13:
1. Yes. 

Vdd is an input to the HPS. For HPS 0.8, Vdd is the voltage difference between pins 1 and 2 of J0.

The HPS 0.8 is designed to receive 3.3V for Vdd.

2. Yes.

For example, with the resistors in the HPS 0.8 parts list, HPS 0.8 will output a voltage near 0 when the hammer is at rest and a voltage of near 2.5V when the hammer is at the stop bar.

Therefore, I selected REF6025 for the ADC reference since 2.5V matches the maximum output of the HPS 0.8.

3. This one is a little more complicated to answer:

REF6030 will also "work" with HPS 0.8. In this case, when HPS 0.8 is at its maximum output of 2.5V, the ADC will output 2.5/3.0 of full scale.  The full scale of ADC is not used.  This gives up some noise margin.  But the piano will "work".

What didn't work was sizing the resistors on HPS 0.8 so that it outputs 3.0V when hammer is at the stop bar. In this case, the voltage drop across the CNY70 was too small (3.3-3.0).

4. Not sure I understand but if the interest is to use REF6030 for all boards, it will also "work" for the HPS 0.8, with the caveat that these sensors will not use the full range of the ADC.

davidedelvento, 2024-07-14:
Thanks, this is extremely helpful, now I understand everything (and I have been able to run some tests).

I now also understand better why you wanted to put potentiometers in some of these designs. There is a key-to-key variability which (upon inspection and testing) is due to the different reflectivity of the hammers. That, compound with the different brightness that can be achieved with different currents at the LEDs make settling to a definitive answer a bit tricky. The QRE1113 can easily reach 3.05V when energized around 3.3V if the target is "shiny" enough (e.g. white paper, new hammers). On the other hand, a less reflective target (e.g. old hammers, even after filing them to make them very white) can stay easily under 2.5V

Puzzlingly, that maximum value does not change when changing the value of the bias resistor on the phototransistor (the non maximum achieved with a old hammer does). As such, I am tempted to have a 6030 installed -- but I fear they would get confused and ship those to Europe and give me the 6025 instead, so I am settling on the 6025 to begin with and then I will figure out some way of "voicing" the hammers to achieve the maximum range.

Thanks