gzweigle/open-hybrid-piano #70: HW: ACE board pins incorrectly set up all on the same side

davidedelvento, 2024-07-04:
In KiCad.

A picture is worth 1000 words

[image]

see how that mismatches the picture of the actual thing which has some things on one side and some on the other

[image]

gzweigle, 2024-07-04:
I don't know how to tell KiCad 3D Viewer Through-hole model Appearance to put on front or back.
Almost done with KiCad v8 update all circuit boards - if I figure out the 3D viewer appearance setting, will include that too.

davidedelvento, 2024-07-04:
> I don't know how to tell KiCad 3D Viewer Through-hole model Appearance to put on front or back.

Just to be clear that is not a matter of appearance in the 3D viewer, it's matter of fabrication error. As is, everything is placed on the top and nothing on the bottom, and if it is sent to fabrication it will come assembled as in the 3D viewer (i.e. incorrectly).

See the `aceA00-bottom.pos` file for detail:

``` 
### Footprint positions - created on 2024-07-04T11:35:51-0600 ###
### Printed by KiCad version 8.0.3
## Unit = mm, Angle = deg.
## Side : bottom
# Ref     Val       Package                PosX       PosY       Rot  Side
## End
```

To select the actual placement, go into the properties of the part (shortcut `E` once the part is selected in the GUI) and pick front or back as placement. Let me see if I can make a PR for this.

davidedelvento, 2024-07-04:
I made #71 trying to fix this issue (hopefully I did not cause other trouble....)

davidedelvento, 2024-07-04:
Same problem with the SCA board, attempted fix in PR #72

gzweigle, 2024-07-04:
> Just to be clear that is not a matter of appearance in the 3D viewer, it's matter of fabrication error

Good point. To fix this I am changing the footprint silkscreen so that it is on the bottom of board. For both SCA and ACE circuit boards. And to do that, I am in the process of revising the ACE from 0.0 to 0.1 and the SCA from 0.0 to 0.1.

I will also change the placement instructions as you note above. For consistency with how I am organizing the GitHub files, these revised PCB will be in new subdirectories which I am creating.

Also, FYI to address KiCad 8 issue I am creating new directories under the original PCB directories, and keeping the original with KiCad 7. I want to keep the original KiCad 7 files in the original directories visible (not buried in github history) because those are what were fabricated and tested... so far.

Thanks!

gzweigle, 2024-07-05:
Because this was a visible board change, I incremented the board version number.
https://github.com/gzweigle/DIY-Grand-Digital-Piano/tree/main/hardware/releases/sca01
https://github.com/gzweigle/DIY-Grand-Digital-Piano/tree/main/hardware/releases/aceA01
I also fixed an unrelated label error on ace version 0.0 and removed the ground plane because it doesn't add much value for this board, but it created a large surface for the soldering to heat which made soldering annoying.
I plan to build and test this board to makes sure I didn't break anything.

closed 2024-07-05 by gzweigle

davidedelvento, 2024-07-05:
> I also fixed an unrelated label error on ace version 0.0 and removed the ground plane because it doesn't add much value for this board

I think the ground plane may add a bit of EMI protection, even though at the frequencies we are dealing with is probably marginal/irrelevant?

gzweigle, 2024-07-05:
Yeah.
Note that the larger SCA and IPS both have ground planes.
For the interposer the runs are very short and the signals relatively slow.
I could make a version (0.2) with the ground plane too if interested.