gzweigle/open-hybrid-piano #74: HW: pedal jack cannot be found

davidedelvento, 2024-07-13:
You wrote "if" in https://github.com/gzweigle/DIY-Grand-Digital-Piano/issues/52#issuecomment-2225571769

Unfortunately, I am unable to find anything suitable and that is stopping progress on https://pianoclack.com/forum/d/1510-diy-electronic-purchase/

Can you please share the part number you used?

gzweigle, 2024-07-13:
Good point that the information in ips20_bill_of_materials_0.txt is not enough.  I will add more details.

Note that I just tried taking the description from the ips20_bill_of_materials_0.txt, and put it into a web search.
"uxcell Audio 6.35mm"
And it returned multiple sources to buy.

davidedelvento, 2024-07-13:
Thanks.

Yes, I did the same search, but unfortunately returned a large number of ebay/amazon/aliexpress results which are not specific product but something without a part number that I could in the order (I did not check them all, but I checked a large number of them and none had a part number). Moreover, many of them had different pin locations which would not match the PCB (as you wrote in that other issue)

I did the same search on Mouser, Digikey etc and that did not return anything. Tweaking the search terms I was able to find something okay (right socket and right number of pins) but none had a footprint compatible with the PCB.

davidedelvento, 2024-07-13:
Maybe this? Not completely clear to me if the footprint is the appropriate one: https://www.mouser.com/ProductDetail/CUI-Devices/SJ-63062A

gzweigle, 2024-07-14:
This in Amazon is it

uxcell 6.35mm 1/4 Inch 3Pin Mono Jack Panel Mount PCB Connectors Metal 6.35mm Headphone Socket Audio Stereo Connector

It is black with red along bottom.

I will put more info in repo too.

davidedelvento, 2024-07-14:
Thanks, but you did not post the link. The first hit is https://www.amazon.com//dp/B00X77SADC/ -- is that the one? It has several bumps in the bottom (probably for slots in the PCB? does it work even if the PCB does not have the slots?)

I will try adding such Amazon part to the order, but given how picky the vendor has been with not exactly part-numbered components (even for pin-headers!!!) I think they will balk at it. In such a case I will order it without and I'll let @AkariProd solder his own ones.

gzweigle, 2024-07-14:
yes that looks right.  Its really easy to solder.

And, yes it works with the slots. Stands the part up off the PCB slightly but the pins are so long they still go through the PCB by a good 1 millimeter.

davidedelvento, 2024-07-14:
Sounds great thanks. As far as I am concerned, this issue could be closed, not closing it myself in case you want to consider including the link in the BOM.

gzweigle, 2024-07-21:
updated documentation

closed 2024-07-21 by gzweigle