gzweigle/open-hybrid-piano #75: HW: pedal resistors clarification

davidedelvento, 2024-07-13:
The BOM contains the lines

```
R23,    4700 ohms,             0805,       ERA-6AED472V,         No,            0.50,        0.125W. 5% tolerance or better. For pedals. Install if using passive pedals.
R25,    4700 ohms,             0805,       ERA-6AED472V,         No,            0.50,        0.125W. 5% tolerance or better. For pedals. Install if using passive pedals.
R28,    4700 ohms,             0805,       ERA-6AED472V,         No,            0.50,        0.125W. 5% tolerance or better. For pedals. Install if using passive pedals.
```

For the group order of https://pianoclack.com/forum/d/1510-diy-electronic-purchase/ some people plan to use passive pedals and some people plan to use active ones (i.e. just CNY70 or QRE1113 sensors like the hammer ones).

Do I understand correctly that if these resistors are installed the active pedals won't work correctly?

Assuming that is the case, consider that there are `16*6=96=88+8` channels, i.e. 8 more than needed for keys only. Given that the pedals use three channels, would it be possible to use 3 channels for the passive pedals and 3 channels for the active ones (and we still have 2 to spare) and avoid this hardware complication?

gzweigle, 2024-07-13:
Will active pedals use the quarter inch jack inputs?

The reason I ask is because:
The quarter inch jacks connect to header J46 and they don't connect to anything else.
The resistors R23, R25, and R28 connect to the quarter inch jack outputs and these resistors don't connect to anything else.
To use quarter inch jacks, one uses a ribbon cable to connect J46 to one of the 96 inputs along the back of the PCB.

If active pedals connect to one of 96 inputs, and do not use the quarter inch jacks, then:
Install the resistors R23, R25, and R28.
And:
(a) Those using passive pedals use the quarter inch jacks.
(b) Those using active pedals connect directly to any of the 96 inputs along back of PCB.

Will that work?

davidedelvento, 2024-07-13:
No active pedals (at least the ones I have in mind) will use the pins like keys.

What you have designed sounds great and would work great, thanks.

Closing the issue.

closed 2024-07-13 by davidedelvento