gzweigle/open-hybrid-piano #77: HW: fuses

davidedelvento, 2024-07-13:
I understand that fuses are just one level of protection and they might or might not work for the purpose, so with all the obvious disclaimers of not taking you liable for anything, this being all at my risk etc etc, I would like to ask your opinion regarding these options (by the way, I would love for the hold and trip currents to be closer, but they aren't)

* For F1 use a 6V, 3.8A (hold), 8A (trip) fuse such as https://www.digikey.com/en/products/detail/bourns-inc/MF-USML380-2/7394484  and for F2 use a 6V, 1.9A (hold), 5A (trip) fuse such as https://www.digikey.com/en/products/detail/littelfuse-inc/MICROSMD190LR-2/3743342 (by the way all SMT fuses seem to be "obsolete" on digikey)
* For both F1 and F2 use the first option above
* For either or both F1 and F2 use a 0 Ohm resistor instead

gzweigle, 2024-07-14:
I clarified the documentation.
We don't know what power supply people will use.

gzweigle, 2024-07-14:
You could leave them off and have individuals decide based on their system.
Its a fairly large 1210 footprint.
That is the direction I changed the documentation.

gzweigle, 2024-07-27:
Added F2 to parts list
https://github.com/gzweigle/DIY-Grand-Digital-Piano/blob/main/hardware/releases/ips20/ips20_bill_of_materials_0.txt

closed 2024-07-27 by gzweigle

gzweigle, 2024-07-27:
Also, I made a video on replacing a surface mount fuse and put the link to video in documentation here:
https://github.com/gzweigle/DIY-Grand-Digital-Piano/tree/main/hardware/releases/ips20