gzweigle/open-hybrid-piano #79: DOC: ribbon cable error

gzweigle, 2024-07-14:
Track down a persistent small coupling error on adjacent notes.
It is the ribbon cables.
Update assembly documentation.

gzweigle, 2024-07-14:
Documentation is updated in stem-piano-g repo

closed 2024-07-14 by gzweigle

davidedelvento, 2024-07-14:
Yeah, ideally the ground and signal cables should be twisted around each other as I described at https://pianoclack.com/forum/d/637-build-your-own-hybrid-piano-step-four-the-cables -- unfortunately that's a pain, but I'm working on building a machine to make that cable-making easier (of course that link is for when I used a different electronics, but still...)

The good news is that there exist pre-twisted cables, which I am still exploring about. The bad news is that it's a niche product with limited options (e.g. length of twisted section, pitch of the cables, etc) which make it hard to adapt to our needs. As I wrote in that page above, one should design the (mechanics of the connectors of the) sensorboards and ADC boards based on cable availability, rather than the other way around... Well, maybe next version.

One example of such cables: https://www.digikey.com/en/products/detail/3m/1700%2F10%2520100/9479266