gzweigle/open-hybrid-piano #87: DOC: make a video of the calibration process

gzweigle, 2024-09-02:
I think a video will make the written documentation clearer.

gzweigle, 2024-09-02:
Add links in GitHub documentation to the two new YouTube videos.
Links are to the video timestamp associated with the GitHub documentation step.
These new videos include extra documentation on construction, testing, and show the complete calibration sequence:
https://www.youtube.com/watch?v=tyN7v7L9VIQ
https://www.youtube.com/watch?v=9pCPfeUCGio

closed 2024-09-02 by gzweigle

davidedelvento, 2024-09-05:
Worth adding to the video the direct link to the relevant part of the manual at https://github.com/stem-piano/stem-piano-g-main/blob/main/documentation/users_manual.md#calibration-hammer-and-damper (which is not very easy to find -- I guess I could make the point for a comprehensive global index, but that would be very tedious so I won't ;) )