gzweigle/open-hybrid-piano #89: HW/DOC: cable crosstalk

davidedelvento, 2024-09-05:
At https://github.com/stem-piano/stem-piano-g-main/blob/main/documentation/component_manual.md#hps-to-ips-connecting-wires you mention the crosstalk across different channels due to use of ribbon cables. I would like to reproduce the problem and test mitigation strategies such as using the twisted cables (which I think I mentioned to you earlier) and others.

To ease my work and help making an apple-to-apple comparison, can you please share the procedure you used to measure this crosstalk (e.g. playing one key and looking at the signal from a nearby channel while everything was on the piano rather than on the table)? If you still have it around, can you please also share the data, so I can check if in my different (as you know) setup I can reproduce a comparable baseline before I try to improve on it?

TIA!

gzweigle, 2024-09-28:
... if only I could remember :)

gzweigle, 2025-04-23:
https://youtu.be/yqkRKSGX6kw?t=229

closed 2025-04-23 by gzweigle