gzweigle/open-hybrid-piano #90: HW: discussion related to IPS 2.1

davidedelvento, 2024-09-08:
To avoid making issue #52 a giant discussion, let me discuss here some things which I think you could/should do for the next version

1. mounting holes -- they are too small, and make the IPS board very flimsy, even just mounting is, especially when attaching/detaching cables and other boards. I know one should not do these things too often, but still... I strongly recommend you make the holes at least M3 and ideally M4 size
2. cabling -- in my brief experience so far, the situation is worse than I thought. Perhaps a mitigation strategy could be to use ubiquitous, inexpensive and high quality ethernet cables, rather than ribbon. Each cable such as https://en.wikipedia.org/wiki/Ethernet_over_twisted_pair would carry signal for 4 sensors. I know your current design involves single sensor boards, but I think you should consolidate them in groups of 4, which is extremely convenient for assembly and regulation (we can have that conversation separately). Even if you want to keep the HSP for single sensors, still grouping the signals in an ethernet cables at the IPS would make things much easier (and a split can be easily made near each sensor if you want to leave them unchanged). This is possible with an interposer board, obviously, but I think it would be ideal if made straight on the IPS board, to minimize size, cost and EMI.
3. In issue #52 you wrote 

> Quarter inch jack interposer board

4. not sure if you mean the same thing I mean for "interposer". For me that word means a PCB that is placed in between two things, both electrically and (most importantly) physically. Here I think the best solution would be a separate board (or even just panel-mount jacks!) with some cabling (ribbon?) going to the IPS, which would have only header pins. Same for most things, including MIDI, Ethernet (should use twisted pairs, but probably the rate is slow enough that it might not matter), power, etc.
5. Related to this, and perhaps you intend to do that but did not say in the list, I think making the IPS much smaller would be better, both for convenience of installation (it is really hard to find that space inside the cabinet without interfering with other things, even more so when using two of them for the damper sensors) and for cost. In my view the IPS should have headers for the Teensy and the SCA, then the multiplexers, and the headers (ideally ethernet-like jacks) for the sensors. Maybe the switches/buttons/LED, and that's it. Everything else should go with few more headers so that it can be installed in a separate location (particularly the display)
6. You also wrote

> Move CAN connector to back of board.

7. I don't understand this, and I strongly oppose it. Dealing with the board as is, it's already difficult. Having headers on the back of the board too would be a nightmare (but perhaps I misunderstood what you meant by "back" and you did not really meant the other side of the board, just a different location?)
8. I think I had a couple of other things, but I don't recall them now and I think they were minor compared to these ones

gzweigle, 2024-09-09:
(2) - agree, a better solution is needed.  The ACE boards help a little but not sufficiently.

(4), (5) - this would be interesting and very flexible... I will think about it. Maybe it is called IPS 3.0 too. Saves at least 1.5 inches of width.
Note that with present board, future SCA design variations can optionally fill the entire TFT PCB space. I designed it this way after considering the mm^2 required for 6 parallel ADC on a new data acq card. So that is why I was ok with the quarter inch jack real estate.

However, taking the modular concept one step further, I could in theory put each function on its own board.
1 board for 96 analog inputs
1 board for data acq
1 board for processing
1 board for display and switches
1 board for pedals
1 board for data i/o
1 board for display
Connect them together with Ethernet cables (as you noted).
Then, everything is highly modular, incredibly flexible/expandable, and minimally small.

davidedelvento, 2024-09-14:
> Note that with present board, future SCA design variations can optionally fill the entire TFT PCB space. I designed it this way after considering the mm^2 required for 6 parallel ADC on a new data acq card. So that is why I was ok with the quarter inch jack real estate.

Would this be to reduce latency / scanning time? If so, rather than having larger SCA and IPS (which are expensive to make) why not have multiple boards like you currently have for hammer and damper sensors? Most PCB houses have a minimum order of 5 boards anyway, so having more boards smaller rather than one board larger makes financial sense (and ease the logistics of cabling since each board could be attached only on (say) an octave or two. Of course a communication protocol better/faster than the CAN bus might be needed.