gzweigle/open-hybrid-piano #93: FW: How to design a MIDI USB device for piano

gzweigle, 2024-09-09:
Location for tracking research.

davidedelvento, 2024-09-09:
These should provide some related info

* https://forum.pjrc.com/index.php?threads/teensy-4-1-adafruit-tinyusb-support.64627/
* https://forum.arduino.cc/t/library-hell-specifically-tinysub-and-midi-in-general/923078/2

and most importantly

* https://www.pjrc.com/teensy/td_midi.html

I've done this on a RPi Pico v1, which is an extremely limited board, so on the Teensy it should not be that hard...

davidedelvento, 2024-09-15:
Looks like this is super-easy, simply

* in the menu Tools -> USB select MIDI + serial rather than only serial
* in line https://github.com/gzweigle/DIY-Grand-Digital-Piano/blob/main/firmware/releases/StemPianoIPS2/src/midiout.cpp#L125 just add `usbMIDI.sendNoteOn(midi_note, velocity_potentially_muted, midi_channel_);`
* and equivalently a couple of lines below with `sendNoteOff`

And should work out of the box. I will try asap (hopefully tomorrow), but I thought to let you know beforehand in case you wanted to try it yourself first.

GalhardoLab, 2024-09-15:
a - it is also necessary to find the instances of CC64, 66, 67 for the pedal messages.

b - there are some arduino midi-usb libraries is which it is possible to buffer several midi commands (maybe up to 16, but I have to double check) and dump them all at once with the same timestamp. This is helpful to send together notes that are triggered within the same scan cycle, thus avoiding a slight "broken chord" effect.

b1 - I am not sure if that arduino library is compatible with Teensy, nor if the teensy library is capable of the same...
b2 - I am not sure if this firmware has one master cycle scan, or if it is interrupt-like... If there is no scan cycle, the hold and dump approach does not make sense.

closed 2024-09-21 by gzweigle

gzweigle, 2024-09-21:
I also edited the documentation.