gzweigle/open-hybrid-piano #96: HW: label inp and gnd on ACE

davidedelvento, 2024-09-14:
Label `gnd` or ground symbol instead of J4 and J9

Label 'inp' instead of J2 and J7

davidedelvento, 2024-09-14:
Or even better label `E` (for emitter) for J2 and J7 and `C` for collector J5 and J10. And now that I see it better, another cabling complication, it would have been better if the rows for E and C were adjacent, as they are it complicates my cable building a lot, I wish I had noticed beforehand.... Maybe for the next version?

gzweigle, 2025-05-04:
Working on damper sensor upgrades and investigated this. The rows match the rows on IPS which helps me keep everything straight.  I will add the idea to swap E and C to the IPS 2.1 list.

gzweigle, 2025-05-17:
> Working on damper sensor upgrades

Sure enough these labels would have been useful.  Adding to an upcoming manufacturing run.

gzweigle, 2025-05-18:
created ACE board version 0.2 with the power, ground, and signal labels.

https://github.com/gzweigle/open-hybrid-piano/tree/main/hardware/releases/aceA02

closed 2025-05-18 by gzweigle