gzweigle/open-hybrid-piano #98: DOC: ethernet connection

davidedelvento, 2024-09-14:
It is unclear how ethernet connection is supposed to happen. I see three options:

* connection to a router with a regular cable (but then what if the router uses DHCP and assigns the board a different IP address than the configured one?)
* connection to the computer with a crossover cable or adapter (see https://en.wikipedia.org/wiki/Ethernet_crossover_cable for details) as it is typically done in these situations (in such a case the hardcoded IP address makes sense)
* connection to the computer with a *regular* cable, assuming that the crossover has happened on the IPS board itself

Any of these solutions are totally fine, but it needs to be documented which one is the appropriate one

gzweigle, 2024-09-28:
Added Ethernet connection documentation.

davidedelvento, 2024-09-28:
Oh, I see so now you send all keys rather than just 12, very nice, given that the bandwidth allow that!

More strictly on my own question, I studied Teensy hardware and software, and it supports *automatic crossover* so you basically connect it to either a switch or a computer directly, and its own firmware (not the one you wrote) interacts with its own hardware and automatically senses which lines are for transmit and which ones are for receive. So you can 

* use any cable (normal or crossed)
* connect to anything (computer or switch) as long as it does *not* use DHCP but honors the IP one hardcodes in your firmware per your instruction

You might want to add this clarification too in the docs

gzweigle, 2024-09-28:
re-opening to remind myself to make this change

gzweigle, 2025-04-26:
Added info in users manual.

closed 2025-04-26 by gzweigle