gzweigle/open-hybrid-piano #100: DOC: make wiring order diagram more prominent

davidedelvento, 2024-09-15:
I was very familiar with https://github.com/stem-piano/stem-piano-g-main/blob/main/documentation/assembly_manual.md#connecting-hammer-sensors existence, but I was not sure of one thing so I wanted to check that.

I've spent 10 minutes finding it. I recommend making it more prominent, perhaps with a dedicated section, and for sure with a "Wiring" link in between 3. and 4. at https://github.com/stem-piano/stem-piano-g-main/blob/main/documentation/construction_manual.md

davidedelvento, 2024-09-28:
Related to this and to make the wiring job easier: would it be possible to change the wiring order in firmware and I mean in a random order?

Reason to ask: given the way that the hammers are grouped, it would be incredibly easier if I could (say) connect the IPS keys numbers 0 to 7 to what are piano keys 2 to 9 and then connect IPS8-15 to piano 10-17 and then IPS16-23 to piano 20-27 etc and then using an arbitrary IPS slot to connect the "loose" piano keys that I missed, in this example so far 0,1,18,19.

I am sure this is *possible*, it's just software obviously. The question is if this is possible by introducing a LUT and using it rather than direct assuming things, and if such a change would be small (handful of places) or massive (turn the whole code base upside down).

Assuming it is "small", would you accept a PR for such a change? I want to make my (and other people's) wiring life easier, but I don't want to diverge from your main code base

gzweigle, 2024-09-29:
I can make this change.  Because I can combine it with void SixChannelAnalog00::ReorderAdcValues(unsigned int *);

gzweigle, 2024-09-29:
Code is updated for custom changing wiring order in firmware.
See hammer_settings.cpp and also the commit diffs.
Let me know if this is what you needed.

Also I updated the wiring diagram documentation

closed 2024-09-29 by gzweigle