gzweigle/open-hybrid-piano #101: FW: damper board no longer compiles

gzweigle, 2024-09-28:
I've made some changes to hammer firmware and need to update the damper firmware.

davidedelvento, 2024-11-28:
Have you had a chance to look at this or not yet? If not, I may give it a stab during the weekend!
Thanks and happy Thanksgiving!

gzweigle, 2024-12-16:
This is still on my to do list.

I previously had the dampers working. But, after changing the mechanical, it presently uses the hammers to estimate the damper positions. This works so well, I have not taken the time to update everything.

davidedelvento, 2025-01-04:
> This is still on my to do list.

Thanks!

> I previously had the dampers working. But, after changing the mechanical, it presently uses the hammers to estimate the damper positions. This works so well, I have not taken the time to update everything.

Yeah but with my hammer (rather than shank) sensors, this is very important for my setup. Life got in the way, so I haven't finished wiring it yet and therefore I haven't even looked at the code either....

gzweigle, 2025-01-07:
ok I updated the code it now compiles.
next I will test it

gzweigle, 2025-04-26:
Updated damper board firmware with all previous and most recent feature enhancements from the hammer board firmware.

closed 2025-04-26 by gzweigle