gzweigle/open-hybrid-piano #102: HW: alternative power supply

davidedelvento, 2024-09-28:
Would you be interested in knowing more, perhaps in a PR, about my power supply strategy which I sketched at

https://pianoclack.com/forum/d/1621-build-your-own-diy-hybrid-piano-step-six-putting-it-all-together/10

I think that has clear benefits compared to your current one, including

* potentially have only two rather than 3 cables per sensor
* avoid the current limiting resistors (if you still have those in your current design)
* simplify wiring, and assembly, especially for people who want to use USB-MIDI, by removing the need for powering up the IPS separately 
* maybe be more stable than your current design (not sure since I have not really studied yours, but I have studied mine :-) )

If you are interested, let me know how is it best for me to share what I have done. If you are not interested, simply close this issue, no offense taken, and if a casual reader cares they will find the link above and will be able to learn from there or contact me separately.

gzweigle, 2024-09-28:
What about putting it here?
https://github.com/stem-piano/sensor-systems

davidedelvento, 2024-09-30:
Sounds great. Let's continue the discussion [there](https://github.com/stem-piano/sensor-systems/pull/2)

closed 2024-09-30 by davidedelvento