gzweigle/open-hybrid-piano #103: Magic damper_velocity_scaling

davidedelvento, 2024-09-28:
https://github.com/gzweigle/DIY-Grand-Digital-Piano/blob/f2202697138da32c7a9a854ba522e4d826138005/firmware/releases/ips2_hammer/hammer_settings.cpp#L141-L144

Does it mean that one (==myself in this case, but anyone in general) needs to adjust this value to make it in that `[0,1]` for my hardware? Two random damper events (ON/OFF) are far from being in that range

```
  key=1 pos_now=0.54 pos_then=0.42 velocity=2.43 threshold=0.50
DetectHammerStrike()  key=1 pos=1.000000 v_max=7.621892 v_inst=8.705567 repc=167 bufp=6
FixedScale(): velocity of key 1 is set at limit, orig velocity was 8.71.
MIDI note (22) index (1) velocity (127) is now ON.
AutoMute: simultaneous = 0 interval = 45596
AutoMute Activated - piano volume reduced for this note.
GetDamperEventData() - damper down
  key=1 pos_now=0.48 pos_then=0.57 velocity=-1.84 threshold=0.50
MIDI note (22) index (1) velocity (82) is now OFF.
AutoMute: simultaneous = 1 interval = 4
```

I understand that you probably meant `[-1,1]` since the damper velocity contains the sign of the direction of motion (you might want to update the comment for that). But even so, I'm well outside of that range with `velocity=-1.84` and `velocity=2.43`, is that going to be a problem?

gzweigle, 2024-09-28:
Looks like with a different hardware setup that the velocity is much larger.
Yes, use the velocity scale so that is in range [-1, 1].

gzweigle, 2024-09-28:
Also I fixed the comment.

closed 2024-09-29 by gzweigle