gzweigle/open-hybrid-piano #104: Use the same nomenclature in the firmware and in the documentation

davidedelvento, 2024-09-28:
I recommend doing this is for everything, but the thing I am struggling now is these switches:

https://github.com/stem-piano/stem-piano-g-main/blob/cfadfc73181cc4308970d7a02c034dba0a3c4c3f/documentation/users_manual.md#ips_sw1_position2-unused

According to my grep'ing there is no `sca_sw1_position2` nor `FREEZE_CAL_VALUES` anywhere in the firmware, so finding where things are happening is harder than it needs to be.

In particular, I have (and cannot) cabled all the keys yet. I notice many problems with my setup and I now suspect that the root cause of at least some is that I have not run (and cannot) the calibration, because I cannot trigger all keys without having all cables. As such, I wanted to change the code of the freeze the calibration and enable that mode anyway even if not all keys are calibrated, so at least I can proceed. I've performed the switch operation, but the LED remains flashing fast, and nothing changes, so perhaps I have not performed it.... Digging into the code and I am confident that I'll have it solved by the time you read this, but I still think a more consistent nomenclature will help those coming after :slightly_smiling_face:

gzweigle, 2024-09-28:
Good point that the documentation does not match the code names.

I added comments to src_hammer.ino.
The src_damper will be the same but I have not checked that in yet because of other compiler errors.

gzweigle, 2024-09-28:
As for calibration - each note calibrates instantly when it is pressed.  The light flashes fast until all 88 are calibrated. But once a note is pressed once, the note is calibrated.

However, the calibration can also get in the way when debugging.  Because it is adding an additional uncontrolled variable to the debugging process.

Here is an idea when trying to get individual keys working:  you could leave sca_sw1_position2 (FREEZE_CAL_VALUES) permanently on. This disables calibration.

Then, either:

- use the velocity scaling setting value to get the velocity in the correct range.

- or, edit six_channel_analog_00.cpp NormalizeAdcValues() and apply a temporary scaling right at the ADC output.  Apply it to normalized_float[].

closed 2024-09-29 by gzweigle