gzweigle/open-hybrid-piano #105: Capture all hammers in software?

davidedelvento, 2024-09-28:
Current version (both release and research, as far as I can tell) of `get_hammer_data.py` still assume 12 notes rather than 96. Since you made that nice video, I assume you already have the code and simply have not pushed it yet.

Related to #97

gzweigle, 2024-09-28:
ah, ok I will fix this.

davidedelvento, 2024-10-14:
Guessing from the update to 

https://github.com/gzweigle/open-hybrid-piano/blob/main/firmware/releases/StemPianoIPS2/src/network.cpp#L67-L82

I tried this patch

```
diff --git a/software/releases/ips2_analytics/get_hammer_data.py b/software/releases/ips2_analytics/get_hammer_data.py
index 9f5d3b1..5b8f0bf 100644
--- a/software/releases/ips2_analytics/get_hammer_data.py
+++ b/software/releases/ips2_analytics/get_hammer_data.py
@@ -25,11 +25,10 @@
 # num_received_values must match number of sent values from the piano.
 
 # Convert encoded integers into original value.
-def convert_to_float(a, b, c):
+def convert_to_float(a, b):
     x  =  int(a)&255
     x |= (int(b)&255)<<8
-    x |= (int(c)&255)<<16
-    y = float(x) / 8388608.0
+    y = float(x) / 32768.0
     if y > 1:
         y -= 2
     return y
@@ -37,14 +36,15 @@ def convert_to_float(a, b, c):
 import socket
 import time
 
-file_length = 20000
-max_packet_size = 100
-num_received_values = 12
+file_length = 40000
+num_received_values = 88
+max_packet_size = num_received_values * 2
+
 
 # This computer's IP address. Must match value in Teensy.
-UDP_IP = FILL IN BEFORE RUNNING
+UDP_IP = "10.10.0.11" 
 # Port number must match value in Teensy.
-UDP_PORT = FILL IN BEFORE RUNNING
+UDP_PORT = 4097
 
 client = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
 client.bind((UDP_IP, UDP_PORT))
@@ -63,7 +63,7 @@ for ind in range(0,file_length):
     # Data is sent as three 8-bit integers. Convert to [-1, ..., 1].
     data_float = []
     for k in range(0,num_received_values):
-      data_float.append(convert_to_float(data[3*k+0], data[3*k+1], data[3*k+2]))
+      data_float.append(convert_to_float(data[2*k+0], data[2*k+1]))
 
     # Write to file as a row of floats.
     # Write only the single note tested.
```

and a small slice of my data looks like 

[image]

Is my patch or my sensing code to be completely broken?

gzweigle, 2024-10-15:
At first glance your patch looks like it should work.

However, just to be sure, I fixed the code. Tested it.  And, put an output file from my testing into the directory with the corrected code.

I created a new issue in case this one needs to stay open for understanding the output data.  However, if the fixed also resolves that, then I will close this issue.

davidedelvento, 2024-10-19:
An update on this. The code you committed is obviously more complete than my quick patch, but the problem I mentioned in my previous update was due to a bug in my plotting script :rofl: and after I fixed that even my earlier capture show decent data (which I still need to study to optimize capture)

I don't think you need to work more on this issue hence I'm closing it

closed 2024-10-19 by davidedelvento