gzweigle/open-hybrid-piano #107: rename repository

gzweigle, 2024-10-10:
The repository name no longer reflects the path this project took.
From GitHub documentation, I can rename the repo and references to the original (including url) will still work.
Those with repo clones may need to reclone or set url.

gzweigle, 2024-10-12:
The repository will be renamed later today.

gzweigle, 2024-10-12:
The new repository name is open-hybrid-piano.

All references to original name automatically link to the new repository name.

For example, https://github.com/gzweigle/DIY-Grand-Digital-Piano/ now redirects to https://github.com/gzweigle/open-hybrid-piano/

closed 2024-10-12 by gzweigle

davidedelvento, 2024-10-12:
Didn't you want to move it to stem-piano organization too?

davidedelvento, 2024-10-12:
FWIW, forks were automatically fixed too (and hopefully clones will do that on their own also, but I'll keep an eye for it).

See https://github.com/davidedelvento/DIY-Grand-Digital-Piano/tree/main (I did not do anything myself)

gzweigle, 2024-10-12:
> Didn't you want to move it to stem-piano organization too?

For now I have been putting the detailed documentation in stem-piano, with links back to open-hybrid-piano (new repo name) for source and issues tracking.

stem-piano I think scales better when we get millions of contributors :) but open-hybrid-piano has all the history.

davidedelvento, 2024-10-12:
My understanding was that you can change the "owner" (from yourself to the stem-piano "organization") keeping all the links and all the history, but in any case, either one works just fine for me :)