gzweigle/open-hybrid-piano #109: DOC: how to proceed in debugging sensor data

davidedelvento, 2024-10-20:
With my sensors and my action (which you know are both very different than yours), I have a problem finding the appropriate calibration. I'm not asking for a fish, but to be taught to fish, which I can then make it into a PR for the documentation, so others can benefit too. I'm describing my problem just as an example: I speculate you went through something similar and so the actual question is "what steps would you suggest performing to debug similar issues". I know some people are mulling the option to use the system with different kind of sensors (maybe not even optical ones), so this would greatly help them too.

It feels like I am stuck at a note-on velocity of almost always 16. See for example this MIDI and corresponding ADC capture

```
note_on channel=1 note=24 velocity=16 time=0
note_off channel=1 note=24 velocity=23 time=0

note_on channel=1 note=24 velocity=16 time=0
note_off channel=1 note=24 velocity=50 time=0

note_on channel=1 note=24 velocity=16 time=0
note_off channel=1 note=24 velocity=84 time=0

note_on channel=1 note=24 velocity=16 time=0
note_off channel=1 note=24 velocity=79 time=0
note_off channel=1 note=24 velocity=4 time=0
note_off channel=1 note=24 velocity=0 time=0
note_off channel=1 note=24 velocity=2 time=0

note_on channel=1 note=24 velocity=16 time=0
note_on channel=1 note=24 velocity=34 time=0
note_off channel=1 note=24 velocity=10 time=0
```

[image]

In this example (which I unfortunately I forgot to separately timestamp on the MIDI capture part, given that real-time MIDI has no inherent timestamps) I had:

* only 5 hammer strikes of which 4 where with a keypress and one was with manual raise of the shank
* note that there are several `note_off` events after the 4th `note_on`. This is likely due to the fact that I grabbed the shank with my fingers and so I moved it in an "unexpected" way
* note that there are 6 `note_on` events, I manually grouped the last two together because that was my "manual" strike in which I did by moving the shank with my finger (no keypress involved), so that may be an actual double-strike. This way I was able to achieve a different-than-16 velocity
* `damper_threshold = 0.22` (otherwise the MIDI channel would be flooded by a continuous stream of `note_off` events)
* `hammer_strike_algorithm = 1`
 
and the following problems:

* velocity is stuck at always 16, regardless of how I make a keypress from ppp to fff
* with manual shank movement (even a relatively slow one, definitely slower than what a keypress achieves) I can get faster velocities, and occasionally a zero (which is different than `note_off` as far as I understand the code in the firmware)
* there are many more peaks in the data captured from the ADC than there are `note_on` events. This is really puzzling

I speculate that the hammers move too fast (my sensors are right under the felt, as compared to yours near the flange) and therefore something in the code is too slow to keep pace and measure it fast enough for achieving velocities faster than 16. Perhaps something else forces it to not be slower than 16 either, and so it's stuck there.... unless I move it with my hand, which is slower enough to be measured and hence other velocity values are generated. Pure speculation here, which I could confirm if there is a quick debug setting somewhere to stop the multiplexing and go "full throttle" on one note only. Is there such an option?

Probably confirming the speculation above, but I tried changing various constants with `_scaling` and `_threshold` and the algorithm (`0` or `1`) without changing anything with regard to this behavior.

gzweigle, 2024-10-20:
I always debugged with calibration off.  One less variable.

Here is some data that could help.  I moved a sensor to be over the hammer.

Attached is:

(1) A picture of the test setup.

(2) Measured values (received over Ethernet) for a very slow (manual move hammer), slow (keypress), and fast (keypress) hammer strike.

(3) A picture of plot of position and velocity.

Observations:

(A) The sensor does not pick up the hammer until it is close to the sensor. So the hammer position data is very nonlinear.

(B) There are less samples due to the faster motion (traveling 40mm at hammer vs 5 mm at flange).  But back-of-the-envelope and looking at this attached data, the 300us sample rate may be ok.

(C) The velocity in attached plot is about 2X for the fast strike compared to the slow strike.  This quick test didn't test if I actually played with 2X the velocity.

[a0_fast_strike.txt](https://github.com/user-attachments/files/17448235/a0_fast_strike.txt)
[a0_manual.txt](https://github.com/user-attachments/files/17448236/a0_manual.txt)
[a0_slow_strike.txt](https://github.com/user-attachments/files/17448237/a0_slow_strike.txt)

[image]
[image]

gzweigle, 2024-10-20:
The results I just shared is with calibration off.

gzweigle, 2024-10-20:
Octave code to recreate my plots.  The zeros in the filter are just to smooth things out.  They aren't necessary.

clear;

x_slow = load("a0_slow_strike.txt");
x_fast = load("a0_fast_strike.txt");

v_slow = filter([1 0 0 0 0 0 0 0 -1],[1],x_slow);
v_fast = filter([1 0 0 0 0 0 0 0 -1],[1],x_fast);

t = [0:length(x_slow)-1]*300e-6;

figure(1);
subplot(2,2,1);
plot(t, x_slow);
grid;
title('position, slow');
xlabel('seconds');

subplot(2,2,3);
plot(t, x_fast);
grid;
title('position, fast');
xlabel('seconds');

subplot(2,2,2);
plot(t, v_slow, 'k', t, v_slow, 'ok');
grid;
title('velocity, slow');
xlabel('seconds');

subplot(2,2,4);
plot(t, v_fast, 'k', t, v_fast, 'ok');
grid;
title('velocity, fast');
xlabel('seconds');

gzweigle, 2024-10-20:
With respect to my observation (B) and this comment in op - "...something in the code is too slow to keep pace and measure them fast enough..."
After thinking about it.
At both hammer and flange, the motion start time and end time are the same.  All that differs is a numerical value.  Therefore, the 300us sample rate is ok for both.  In other words, the "code speed" is ok for measuring at hammer.  The problem is somewhere else.

gzweigle, 2024-10-20:
OK, I was able to get the piano to work, using the test setup (picture in previous comment) with sensor over hammer.

The hammer travels approximately 40mm. The CNY70 and QRE1113 cover a 10X collector current range starting at approximately 5 - 7 mm. This is from my understanding of the datasheets.

Here are the new settings values (hammer_settings.cpp file):

I changed the hammer algorithm to threshold. Maybe not required. But simplified getting it to work.
hammer_strike_algorithm = 1;

Then, I moved the threshold down significantly. Because of the 40 mm vs "5 mm" ranges.
strike_threshold = 0.7;

Also, changed the velocity scaling to a smaller value. This is because of the large nonlinearity when the hammer "comes into view" of the sensor:
velocity_scale = 0.10;

To get the note to sustain, I had to set the threshold quite small, because the catch position is outside the sensor normal range. 
damper_threshold = 0.06;

I used the Ethernet streamed data to help me set the above values.

Here is the results for playing A0 soft and loud:

MIDI note (21) index (0) velocity (90) is now ON.
MIDI note (21) index (0) velocity (0) is now OFF.
MIDI note (21) index (0) velocity (66) is now ON.
MIDI note (21) index (0) velocity (0) is now OFF.
MIDI note (21) index (0) velocity (78) is now ON.
MIDI note (21) index (0) velocity (0) is now OFF.
MIDI note (21) index (0) velocity (50) is now ON.
MIDI note (21) index (0) velocity (0) is now OFF.
MIDI note (21) index (0) velocity (97) is now ON.
MIDI note (21) index (0) velocity (0) is now OFF.
MIDI note (21) index (0) velocity (107) is now ON.
MIDI note (21) index (0) velocity (0) is now OFF.
MIDI note (21) index (0) velocity (23) is now ON.
MIDI note (21) index (0) velocity (0) is now OFF.

The white sticker does not want to stay on the felt hammer! That was the biggest problem.

gzweigle, 2024-10-20:
For tests above, I turned calibration off by setting switch sca_sw1_position1 to the ON position.
Because of 40 mm vs "5 mm" the calibration code may need changes. So I didn't want it confusing matters.
The piano will play without calibration. Calibration makes the piano much more accurate but is not required for basic playing.

davidedelvento, 2024-10-20:
Thanks Greg for this wealth of information. I am starting this test right now!

FWIW, I was doing almost the same things you are, my major difference was

```
strike_threshold = 0.7;
velocity_scale = 0.10;
```

which instead I had to values closer to default. I'll let you know shortly where I land!

davidedelvento, 2024-10-20:
Velocity scale was the main culprit, looks like now I am in business, and the piano is somewhat working. I will play with the various settings and in auto-calibration soon and see how things go.

I guess part of my confusion when exploring this alone, was that I was thinking of the `velocity_scale` as something relevant to MIDI and not to the ADC data which I expected to see similar to the following regardless of the `velocity_scale` value. I need to fully understand what that does exactly and maybe we need to document it for others

[image]

There are two possible problems that I still see from this picture:

1) the ripples (not very visible with this scale but always there) which are especially clear if there is no hammerstrike. This is not trivial 60Hz light noise because I've tested it in the dark, however it can be one of the plenty LEDs on the IPS board, e.g. the Ethernet ones which are actually pointing toward the direction of the note I'm testing. I will try some screening but to help debugging I need confirmation of the duration of each line in the captured `hammer_position.txt` file: is that 300us?

2) I may be clipping at the closest hammer-sensor distances which (if true) means I should change the resistor on the ACE board :disappointed:

gzweigle, 2024-10-20:
(1) "to help debugging I need confirmation of the duration of each line"

Yes, each row in the file is one set of samples and the sample period is 300 microseconds.

The sample period is set by the adc_sample_period_microseconds value in hammer_settings.cpp.

The sample period value must be larger than the time it takes for all processing in the Teensy.  Oscilloscope testing (using a test point on the board) I've measured the Teensy processing to take approximately 250 microseconds.  So, I used a 300 microsecond sample interval to give a little margin.

If the codebase ever needs large changes, that could necessitate changing the sample rate.

There are no restrictions on making the sample period longer (sample rate slower), other than it makes the velocity estimates less accurate.  I have occasionally used longer sample periods when I had to add a large amount of test code for debugging things.  This is something to be aware of.  If you ever add a lot of test code and suddenly see really strange behavior, try increasing the sample period temporarily to something like 400 microseconds.

I've also tested with shorter sample periods (much faster sample rates) by removing some of the "unnecessary" Teensy code.

davidedelvento, 2024-10-20:
Thanks again!

That noise is indeed at 60Hz, so it must be coming somewhere from the AC distribution frequency. I tried a few other things, including routing the AC power supply elsewhere (as I wrote, this is already with lights off and inside an enclosure, so the only possible luminal noise would be the LEDs on the IPS). I also wired one octave using the Ethernet cables for most of the distance (of course there were still short jumpers, but I twisted them), and the noise remained the same. Here is a close-up

[image]

The waveform of this noise is peculiar and I think that is telling me something, so I did the FFT and it's telling me (as obvious from the time domain plot too) that there are many other frequencies, but most importantly (which was not obvious from the time domain plot) that the vast majority of the noise is coming from the harmonics of the 60Hz

[image]

Cleaning up the FFT spectrum removing everything under `2e-5` and performing the inverse creates almost exactly the noise from the time domain, so there is nothing else. I guess the next thing I will try is to battery power the LEDs so I can distinguish LED noise vs sensor/ADC noise (I could do that with an oscilloscope, but I don't have a portable one and the instrument is in a cabinet in a bedroom, so batteries are more practical)

gzweigle, 2025-05-04:
Did this get resolved?
If so, we can close the issue.

davidedelvento, 2025-05-04:
Did it get resolved? Yes, maybe and no.

Yes: if I power everything from batteries, this problem is gone.
Maybe: if I power everything from an excellent power source (e.g. a linear power supply), I can probably still see this issue, but I need to look pretty hard to find it.
No: if I power the LEDs from an excellent linear power supply and the rest of the board from the USB, the problem is still there. Among the ones I have tried, some computers have really bad spectrum (e.g. cheap, old, laptop), and some others are decent (but I would not call them acceptable, e.g. old iMac).

So in conclusion, I think you should do either (or both) of the following:

* mention this problem somewhere in the documentation, warning people to use the best power supply they can afford (and avoid USB-powering at all costs, which I know at least two other people on pianoclack are instead interested in exploring)
* see if you can change anything in the power circuit (of the digital) to make this problem moot. The easy/obvious way would be to add some large capacitor to make a low pass filter. Depending on the power draw that might or might not be practical, I'll leave it to you to check what is the case -- taking also in mind that large electrolytic capacitors have a finite lifespan which may cause problems down the road... 

Feel free to close it if you plan to do no action, or to keep it open as a reminder

gzweigle, 2025-05-04:
Interesting...

During earlier development I would often run the piano with USB power. Although never with all 88 keys (only with a subset of sensors, that USB could handle without exceeding its limit).

For example, note here the power input isn't soldered, only USB.
https://youtu.be/DxdRO3M7cgg?t=227

For reference, here is the existing power supply documentation:
https://github.com/stem-piano/stem-piano-g-main/blob/main/documentation/component_manual.md#external-5-volt-power-supply
I could put the warning in this section.

gzweigle, 2025-05-04:
Added documentation at link above on power supply noise.
Also noted what power supply conditioning is presently on the PCBs, for example the LDOs.

closed 2025-05-04 by gzweigle