gzweigle/open-hybrid-piano #111: Main board fails to initialize if pedals are in some circumstances

davidedelvento, 2025-01-27:
I finally managed to wire the pedals (which given the geometry of my setup was very challenging). For all three pedals, I am using https://www.sparkfun.com/sparkfun-line-sensor-breakout-qre1113-analog.html as pedal sensor.

I have all three pedals connected to the main board, the board does not initialize, does not present itself over USB-MIDI and does not even connect to Ethernet. If I disconnect two pedals and leave just one, it still behaves in this anomalous way. 

However, in the single pedal case ONLY, the main board behaves correctly if before starting the boot, I keep the pedal down (which mean keeping the distance from the sensor high). I cannot reproduce this correct behavior with two or three pedals connected, regardless of how I keep the pedals during the boot.

I am sure there must be something wrong with either my wiring or my sensors, but I think that this can still be considered a firmware bug since even with incorrect data coming from the sensors the board should boot correctly (and help me debug what is wrong, ideally, e.g. by dumping over ethernet the wrong values of the pedals), hence I'm reporting this here.

As a side note, from this behavior, do you have any clue about what might be wrong?

gzweigle, 2025-01-27:
You may want to try putting a

Serial.println("got here");

after each function call in setup()

(and loop() if the code completes setup() ok).

To find where the code hangs.

gzweigle, 2025-01-27:
Also, here is what I put in the help file for similar symptoms.  May give ideas for places to look:

https://github.com/stem-piano/stem-piano-top/blob/main/HELP.md#the-piano-does-not-work-at-all

gzweigle, 2025-04-26:
Close based on adding further help as described above.
Can reopen if issue persists.

closed 2025-04-26 by gzweigle