gzweigle/open-hybrid-piano #132: FW: yet another accuracy research

gzweigle, 2025-06-06:
When playing piano, still not 100% happy with the accuracy.

Also discussed here:
https://github.com/gzweigle/open-hybrid-piano/issues/14
https://github.com/gzweigle/open-hybrid-piano/issues/55
https://github.com/gzweigle/open-hybrid-piano/issues/63

And here:
https://www.youtube.com/watch?v=TemXGmkoc-g
https://www.youtube.com/watch?v=yqkRKSGX6kw

davidedelvento, 2025-06-22:
Thanks for sharing. Have you redone the mechanical structure of the sensors as you suggested you were doing in one of your updates?

I think using the cheap L extrusions from hardware store is one (major?) source of your problems since they are not rigid enough and prone to bending and vibrating, let alone being accurate in the initial position. I think 4040 extrusions (like the ones I am using) are a solution to this part of the problem.

Also, having your sensors on the shank, IMHO complicates the matters since there is possibly a note-to-note variation and calibration needed. I think my design of having the sensors on the hammers, with a set spacer with a small hole dramatically improves this problem too, since now each note is exactly sensed then the hammer hits the spacer and the sensor notice it from the small hole (remember, I use small sensors, not the huge CNY70). 

I will still have a problem with possibly different reflectivity from note to note, due to difference in hammer felt, but I think I can solve that by filing, which is pretty easy to do (even if a bit messy, but my design allows removal of the action from the cabinet, so the electronics is not affect by that).

To be completely open, none of these are my inventions, I've just taken them from Jay, like I took the electronics and firmware from you. Since you're going faster than I (I'm still stuck at cabling) I wonder if you tested these other options and reached any conclusions.

gzweigle, 2025-06-22:
Thanks. Cool to see a different approaches in the works.

One of my design requirements is sampled position measurements, hammer and damper. So I need the sensed surface to stay less than 8 - 10 millimeters from the sensor. And closer than that is better. Hence shank.

For accuracy, I am working on an optional plug-in addition (to existing hardware system) reference the digital calibration algorithms.

My goal went from "make an open source hybrid piano that works", to "make an open source hybrid piano with high precision and accuracy".

gzweigle, 2025-06-28:
Here is a writeup.

It seems to work. Will make a video when get time someday.

https://github.com/gzweigle/hybrid-piano-calibration/tree/main/pdps

gzweigle, 2025-06-28:
Some browsers are better than others at displaying the equations and equation numbers in document linked above.